beepnl/BEEP #7: Handling email addresses with + characters

amotl, 2019-06-10:
Dear @pvgennip,

just a quick note. When registering for an account with an email address of `andreas+t1@...-`, we found that the form-prefill back on the login page you are redirected to after confirming the activation email would not honor plus characters (+) properly.

This yielded 
[image]
to us.

Just a convenience thing, not really a stopper bug.

Cheers,
Andreas.

amotl, 2019-09-21:
The same thing happens on the password recovery dialog.

[image]

pvgennip, 2019-12-17:
Sorry for late resolution, will be solved in next version (this month)

closed 2019-12-17 by pvgennip