beepnl/BEEP #15: Force https in .htaccess

joek, 2020-02-13:
Hi,
I am pretty new to BEEP but I am super excited about the project as I had the idea of an open source hive management solution too but no time to dig into. 

Currently I am working on a docker-compose deployment to simplify the deployment on a private server and local development (for this part some work is still needed).

During this process I faced one issue which is hard to automate.

 
**Problem:**
The force https section in the `.htaccess` file is preventing to run a none https deployment in a local setup. The result is, I have to adjust this file every time I am checking out the project. 

**Solution proposal:**
If the force https part is part of the `apache/beep-public.conf` we can avoid having this inside the `.htaccess`.

pvgennip, 2020-02-20:
Yes, good idea. You can create a fork and do a pull request to change this.

pvgennip, 2020-05-14:
Sorry for the very slow merging, but thanks again!

closed 2020-05-14 by pvgennip