beepnl/BEEP #21: Change response for collection endpoints in case of empty collections

georgevanvliet, 2020-02-23:
Please have collection endpoints, such as `GET /api/devices` return an empty array if the collection is empty, and make sure the http response code is also 200 OK.

Also have this behavior when applying query strings to collection endpoints e.g. `GET /api/devices?hardware_id`.

Currently endpoints, such as the one above, throw a 404 code and a string. This makes the client deserialization harder than necessary (invalid json, exception branching).

pvgennip, 2020-05-14:
This will break the current web app, because that depends on empty results having response code 404, so this would need a thorough refactoring.

georgevanvliet, 2020-05-14:
Ok. I currently use custom error handling for all collection endpoints. It converts a 404 on collection endpoints to an empty array. It's dirty but it works for now.

closed 2020-07-03 by pvgennip

maartenvandillen, 2022-05-31:
I agree that this needs to be fixed. 404 means resource not found, but an empty collection is still a valid resource.