beepnl/BEEP #24: Throw a 401 error when using no or expired token when using content/type application/json

georgevanvliet, 2020-02-25:
Currently, when consuming the api as a JSON client, with content-type set to application/json, a 302 redirect is thrown when using none or an expired/invalid token. There is no way for the client to detect that its token has expired. Other than having a 302 imply an authentication failure and grep the body for `login` :(

Please throw 401 errors when the token is not present or invalid, and the content-type is anything other than html, e.g. application/json.

Also see #14 for further improvements. Currently there is no way to derive the ttl of the token and we need to wait for a call to fail, before we present the login screen.

pvgennip, 2020-02-26:
This is solved by adding the header **Accept : application/json**

pvgennip, 2020-02-26:
Default behaviour is to redirect to the login screen

closed 2020-02-26 by pvgennip