beepnl/BEEP #27: Add proxy endpoints for TTN

georgevanvliet, 2020-03-06:
Please add two endpoints that can be used for proxying to the TTN application manager:
* A `GET` for TTN devices
* A `POST` for TTN devices

The post receives the following payload:
```
{
  "dev_id": "<deviceId>",
  "lorawan_device": {
    "app_id": "beep-digital-hive-monitoring",
    "dev_eui": "devEui", 
    "dev_id": "deviceId", 
    "app_key": "appKey",
    "app_eui": "70B3D57ED0028D38" 
  }
}
```

pvgennip, 2020-03-23:
Does the POST to TTN need to have the payload, or does the endpoint for POSTing to TTN (on the BEEP API) receive this payload?

georgevanvliet, 2020-03-24:
Both. The devEui, appkey and deviceid are provided by the client. These should be used to (re)construct the above payload and pass it to ttn. Beep api should enforce proper appeui.

pvgennip, 2020-03-24:
@georgevanvliet How can the client (app in this case) have knowledge of the <appKey>, shouldn't this be also enforced by the BEEP API settings?

pvgennip, 2020-03-24:
This API calls can be tested (https://test.beep.nl/docs/#apidevicesttnhardwareid-get):
`GET api/devices/ttn/{hardware_id}`
`POST api/devices/ttn/{hardware_id}/dev_eui/{dev_eui}`

georgevanvliet, 2020-03-27:
Ok works!

closed 2020-03-27 by georgevanvliet