beepnl/BEEP #42: TTN decoder yields wrong hardware_version and hardware_id?

avbentem, 2020-06-30:
Just looking at the code of [ttn-payload-decoder.js](https://github.com/beepnl/BEEP/blob/master/ttn-payload-decoder.js) the decoder, I **think** it has at least two errors for `hardware_version` and one for `hardware_id`.

The test payload of `0100010003000402935685E6FFFF94540E01237A26A67D24D8EE1D000001` on port 2 currently yields `"37718.34278 ID:16749907"` and `"01237a26a67d24d8ee"` but I think those should read `"168790.34278 ID:4294939732"` and `"0e01237a26a67d24d8ee"`.

## hardware_version is missing first byte?

```javascript
decoded.firmware_version = ... + bytes[6]);
decoded.hardware_version = ((bytes[8] << 8) + bytes[9]) + ...
```

The code skips `bytes[7]`. Also, the code seems to document that's actually part of this `hw version` attribute:

```javascript
// 0100010003000402935685E6FFFF94540E01237A26A67D24D8EE1D000001
// 01 00 01 00 03 00 04 02 93 56 85 E6 FF FF 94 54 0E 01 23 7A 26 A6 7D 24 D8 EE 1D 00 00 01 
// 0  1  2  3  4  5  6  7  8  9  10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 
//    pl  fw version    hw version                 ATTEC ID (14)                 app config
```

Same goes for other parts of the decoder, skipping `bytes[9]`:

```javascript
decoded.firmware_version = ... + bytes[8]);
decoded.hardware_version = ((bytes[10] << 8) + ... 
```

## hardware_version is not handling the MSB in "ID" correctly?

Note that when using bitwise operators in a TTN decoder, [JavaScript expects _signed_ 32 bits numbers](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Bitwise_Operators#Signed_32-bit_integers) using [two's complement encoding](https://en.wikipedia.org/wiki/Two%27s_complement), hence allowing for numbers from -2,147,483,648 through 2,147,483,647.

Knowing that, the following smells:

```javascript
... + " ID:" + ((bytes[12] << 32) + (bytes[13] << 16) + (bytes[14] << 8) + bytes[15]);
```

First, I'm quite sure the `<< 32` should read `<< 24`? As JavaScript is working with 32 bits values, using `<< 32` effectively is the same as not shifting at all: `0xFF << 32` is actually 0x000000FF, and `0xFF << 40` is 0x0000FF00, and so on. Unlikely that's what you want?

Second, given that JavaScript expects two's complement encoding, using `<< 24` will yield a negative number if `bytes[12]` has its leftmost bit set. (So: if the value is 0x80 or above, or if `bytes[12] & 0x80` is not zero.)

When using `<< 24` rather than `<< 32` then you can test with the payload you're already showing in the code, which holds `FFFF9454` in `0100010003000402935685E6FFFF94540E01237A26A67D24D8EE1D000001` on port 2.

This can be circumvented by using [the zero-fill right shift `>>>`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Bitwise_Operators) along with actually shifting zero bits, or by not using bitwise operators at all. Any of the following will work:

```javascript
... + " ID:" + ((bytes[12] << 24 >> 0) + (bytes[13] << 16) + (bytes[14] << 8) + bytes[15]);

... + " ID:" + (((bytes[12] << 24) + (bytes[13] << 16) + (bytes[14] << 8) + bytes[15]) >>> 0);

... + " ID:" + (bytes[12] * 0x1000000 + bytes[13] * 0x10000 + bytes[14] * 0x100 + bytes[15]);
```

Aside, when using bitwise OR rather than addition, one needs fewer parentheses, as addition [has a higher precedence](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Operator_Precedence) than the shift operator:

```javascript
... + " ID:" + ((bytes[12] << 24 | bytes[13] << 16 | bytes[14] << 8 | bytes[15]) >>> 0);
```

Better yet, just define a helper that takes an array of bytes and returns an unsigned integer:

```javascript
// Convert the array of bytes to an unsigned integer, MSB.
// This is only safe up to 0x1FFFFFFFFFFFFF, so: 6 bytes.
function toUInt(bytes) {
  return bytes.reduce(function(acc, b) {
    // Unsigned, so don't use any bitwise operators, which would always
    // yield a signed 32 bits integer instead.
    return acc * 0x100 + b;
  }, 0);
}
```

With another helper to accept a full array of bytes and a start index and length:

```javascript
function getUInt(bytes, start, len) {
  return toUInt(bytes.slice(start, start + len));
}
```

...one could then use:

```javascript
decoded.hardware_version = getUInt(bytes, 7, 3) + "." + getUInt(bytes, 10, 2) + " ID:" + getUInt(bytes, 12, 4)
```

## hardware_id is missing first byte?

The code skips `byte[16]`. The documentation in the code suggests it's part of `hardware_id`, named `ATTEC ID` in the example. 

```javascript
decoded.hardware_version = ... + bytes[15]);
decoded.hardware_id      = toHexString(bytes[17], 2) + toHexString(bytes[18], 2) + toHexString(bytes[19], 2)
                         + toHexString(bytes[20], 2) + toHexString(bytes[21], 2) + toHexString(bytes[22], 2)
                         + toHexString(bytes[23], 2) + toHexString(bytes[24], 2) + toHexString(bytes[25], 2);
```

Also, adding a helper that takes an array to print as hex keeps the code short:

```javascript
// Get a hexadecimal representation with leading zeroes for each byte.
function toHex(bytes) {
  return bytes.reduce(function(acc, byte) {
    return acc + ('0' + byte.toString(16)).slice(-2);
  }, '');
}

...

decoded.hardware_id = toHex(bytes.slice(16, 26));
```

## Bonus

It seems lines like the following are meant to convert fewer than 4 bytes into a signed integer?

```javascript
decoded.t_i = hexToInt(toHexString(bytes[ds18b20_values_start_byte], 2) 
            + toHexString(bytes[ds18b20_values_start_byte+1], 2), 4); 
```

If true, then [sign extension](https://en.wikipedia.org/wiki/Sign_extension#:~:text=Sign%20extension%20(abbreviated%20as%20sext,positive%2Fnegative)%20and%20value.) is your friend:

```javascript
// Sign-extend a 16 bits number to 32 bits to make JavaScript understand negative values, by
// shifting the MSB 24 bits to the left, followed by a sign-propagating right shift of 16 bits
// to effectively shift 8 bits.
bits.decoded.t_i = bytes[ds18b20_values_start_byte] << 24 >> 16 | bytes[ds18b20_values_start_byte+1];

```

Of course, a helper might be better:

```javascript

// Get a signed integer from 2 bytes, MSB.
function toInt16(bytes) {
  // Sign-extend to 32 bits to support negative values: shift 16 bytes
  // too far to the left, followed by sign-propagating right shift
  return bytes[0] << 24 >> 16 | bytes[1];
}

function getInt16(bytes, start) {
  return toInt16(bytes.slice(start, start + 2));
}

...

bits.decoded.t_i = getInt16(bytes, ds18b20_values_start_byte);

```

Enjoy!

pvgennip, 2020-07-03:
Thanks a lot for you comment and suggestions.

The **hardware_version** and **hardware_id** are correct as they are in the script. They should exclude the 'category' byte that indicates that the start of version and id is after respectively 02 and 0E (see the BEEP base firmware manual: https://github.com/beepnl/measurement-system-v3/blob/master/firmware/BEEP%20base%20-%20ID190222-02%20-%20Firmware%20-%20English.pdf)

The suggested bonus improvement, I will definitely use!

closed 2020-07-03 by pvgennip

avbentem, 2020-07-03:
Okay, but skipping 02 and 0E only explains 2 out of the 3 reported issues? I really cannot see how `<< 32` would ever be what's expected?

avbentem, 2020-07-13:
@pvgennip did you see my comment? (I have no idea if a correct `ID:` part in `hardware_version` is even important to you, but it surely looks wrong now.)

pvgennip, 2020-07-13:
@avbentem Yes I did, sorry for my late reply. Thank you for pointing out **hardware_version is not handling the MSB in "ID" correctly?**. The hardware_version has not yet been important because wo only have one at the moment :-). But it will be in the second batch later this year.

I will also implement that for all properties including bootcount a.s.a.p. 

Thanks!