beepnl/BEEP #43: Payload converter will not be supported in next release of TTN

avbentem, 2020-06-30:
The next version of the TTN LoRaWAN Stack, V3, is due on the community network somewhere this year. As per https://www.thethingsnetwork.org/forum/t/convert-custom-payload-to-lpp-payload-in-ttn-console/12171/9:

> We’re dropping the converter and validator in V3 as they’re not being used; you can put conversion in the decoder, and it makes more sense to drop/filter out invalid values higher upstream.

So, it might be better to be prepared, and combine [ttn-payload-decoder.js](https://github.com/beepnl/BEEP/blob/master/ttn-payload-decoder.js) and [ttn-payload-converter.js](https://github.com/beepnl/BEEP/blob/master/ttn-payload-converter.js) into a single Decoder.

Better yet, maybe decoding should be done in the BEEP backend?

pvgennip, 2020-07-03:
Thanks for this remark. We already have the converter and decoder in place in the BEEP backend, so we will switch to that soon.

pvgennip, 2020-08-05:
Rolled out at live

closed 2020-08-05 by pvgennip