beepnl/BEEP #48: Instructions - Installation server

majo91, 2020-07-26:
Hey!

I'm impressed by the system which you developed and i would like to use it with my own server but i have problems with the installation of it.

Do you have instructions which works or even a Git container which works? For example the webapp directory is not existing in the public folder.
I used the master branch on a new Debian 9 machine.

Best Regards

pvgennip, 2020-08-04:
@majo91 correct, the **webapp** directory is not a physical directory, but is a route created by the backend system. This is so the php blade template can be used to create the homepage of the app.

Can you post the details of the problems that you run into?

majo91, 2020-08-04:
I had to install nearly everthing manually (apache2, mysql, php, etc). Additionally to get no errors with the run_action.sh file (step 3) i had to install redis-server and changed the database.php in the config folder.
Well, ok the virtualhosts seems not working - maxbe i made i mistake but i want to open my page without the virtualhost i get redirected to beep.nl
i can't find a mistake, i made all necessary changes according to the instructions

majo91, 2020-08-05:
I have installed everything manually on a fresh server (Apache, MySQL, PHP 7.3, ..) Then I followed the instructions and at step 3 I already had the first problems. I had to adjust the database.php file because of an error message (I had to remove the mode "NO_AUTO_CREATE_USER"). I had to install the redis-server.
Then I set up the virtualhosts that work.
Now when I want to call my page api.mares3d.com/webapp#!/login/create I get the message "not found".
If I only enter api.mares3d.com, I will be forwarded to beep.nl. With the address app.mares3d.com I also get the message "not found".

pvgennip, 2020-08-05:
@majo91 Can you try to go to api.mares3d.com/login and log in with an admin user?

pvgennip, 2020-08-05:
And did you also install the .htaccess file (and enable that in Apache), that does the routing?

pvgennip, 2020-08-05:
Did you run ./run_actions.sh to install all composer packages and make the required directories accessible?

majo91, 2020-08-05:
> @majo91 Can you try to go to api.mares3d.com/login and log in with an admin user?

Page Not found

majo91, 2020-08-05:
> Did you run ./run_actions.sh to install all composer packages and make the required directories accessible?

Yes i did run the run_action.sh and the required directories should be accessible i think

majo91, 2020-08-05:
> And did you also install the .htaccess file in the /public directory (and enable that in Apache), that does the routing?

There is no description in the manual. there is an .htaccess file in the public folder. how to install or enable that in apache?

pvgennip, 2020-08-05:
> > And did you also install the .htaccess file in the /public directory (and enable that in Apache), that does the routing?
> 
> There is no description in the manual. there is an .htaccess file in the public folder. how to install or enable that in apache?

sudo a2enmod rewrite

majo91, 2020-08-05:
> > > And did you also install the .htaccess file in the /public directory (and enable that in Apache), that does the routing?
> > 
> > 
> > There is no description in the manual. there is an .htaccess file in the public folder. how to install or enable that in apache?
> 
> sudo a2enmod rewrite

thanks - now i get the login page

pvgennip, 2020-08-05:
> /var/www/bee/public/storage

Is seems like /var/www/bee/public/storage is not there yet. Create it with:
mkdir /var/www/bee/public/storage
sudo chmod 777

majo91, 2020-08-05:
now if i type in the information for creating a new user, it seems that it 

> > /var/www/bee/public/storage
> 
> Is seems like /var/www/bee/public/storage is not there yet. Create it with:
> mkdir /var/www/bee/public/storage
> sudo chmod 777

root@localhost:/var/www/bee/public# mkdir storage
mkdir: cannot create directory ‘storage’: File exists

pvgennip, 2020-08-05:
> > > /var/www/bee/public/storage
> > 
> > 
> > Is seems like /var/www/bee/public/storage is not there yet. Create it with:
> > mkdir /var/www/bee/public/storage
> > sudo chmod 777
> 
> root@localhost:/var/www/bee/public# mkdir storage
> mkdir: cannot create directory ‘storage’: File exists

Ok, perfect :-)

majo91, 2020-08-05:
if i want to create a user, it does nothing when i confirm

pvgennip, 2020-08-05:
> if i want to create a user, it does nothing when i confirm

You should set up the right e-mail server specs in the .env file

majo91, 2020-08-05:
> > if i want to create a user, it does nothing when i confirm
> 
> You should set up the right e-mail server specs in the .env file. You you can create an e-mail account for this at your server.

i did, it should work with these settings

MAIL_DRIVER=smtp
MAIL_HOST=smtp.ionos.de
MAIL_PORT=587
MAIL_USERNAME=josef@mares3d.at
MAIL_PASSWORD=[PASSWORD]
MAIL_ENCRYPTION=ssl
MAIL_FROM_ADDRESS=josef@mares3d.at
MAIL_FROM_NAME=Josef

pvgennip, 2020-08-05:
> if i want to create a user, it does nothing when i confirm

Where do you try to create a new user? at /webapp ?

majo91, 2020-08-05:
> > if i want to create a user, it does nothing when i confirm
> 
> Where do you try to create a new user? at /webapp ? Please check your database if a user has been created.

here: http://api.mares3d.com/webapp#!/login/create
the table users is still empty

pvgennip, 2020-08-05:
> > > if i want to create a user, it does nothing when i confirm
> > 
> > 
> > Where do you try to create a new user? at /webapp ? Please check your database if a user has been created.
> 
> here: http://api.mares3d.com/webapp#!/login/create
> the table users is still empty

It should be at https:// otherwise you will have to disable the https redirect in the Apache config

majo91, 2020-08-05:
ok i think i works now.
i have to check the mail settings with my provider because the "verify Email Address" mail get blocked

majo91, 2020-08-06:
Thank you! now it works!

closed 2020-08-06 by majo91