beepnl/BEEP #49: Include hive_layers as an argument for creating hives when saving location

JuliaBD, 2020-07-30:
**Issue:** Creating a new apiary gives the error that only 14 out of 15 arguments are provided for the createHive call on line 293 of HiveFactory.php, inside createMultipleHives function. The 'hive_layers' argument (see end of line 15) is missing. (The result is that new location is created in the db, but without any hives or hive_layers)

**Solution:** Please update 'createMultipleHives' function (in HiveFactory.php, called in LocationController.php) to include hive_layers as an argument.

-in LocationController, please call the property 'layers' instead of 'hive_layers' (better for my code consistency)
-ignore 'brood_layers' and 'honey_layers' properties whenever 'layers' is not null? Or whenever 'brood_layers' and 'honey_layers' are null / undefined?

pvgennip, 2020-08-04:
Deployed on test.beep.nl

pvgennip, 2020-08-05:
Deployed at live

closed 2020-08-05 by pvgennip