beepnl/BEEP #68: Adding weather data

amotl, 2020-11-24:
Dear Pim,

I just became aware of #67 and want to humbly make you aware about what @gutzbenj, @wetterfrosch and all other contributors are conceiving over at https://github.com/earthobservations/wetterdienst.

While this is currently just about ingesting weather information from @DeutscherWetterdienst, we want to tap into data from other domestic weather agencies soon.

So, maybe you can share some information about where we might want to look into when approaching open weather data from the Dutch weather agency?

With kind regards,
Andreas.

amotl, 2020-11-24:
We see at https://github.com/beepnl/BEEP/blob/11d6716b5451ee3747a22741a5c0f643bf772a0e/app/Http/Controllers/ResearchController.php#L413-L424 that you are going to InfluxDB for acquiring weather information. This aligns well with the InfluxDB feature support we have been bringing in with [version 0.8.0](https://wetterdienst.readthedocs.io/en/latest/pages/changelog.html#id4) and are currently refining with the [most recent improvements to the development head](https://wetterdienst.readthedocs.io/en/latest/pages/changelog.html#development).

amotl, 2020-11-24:
Hi again,

quickly answering myself here, I found the following information for acquiring data from the [Royal Netherlands Meteorological Institute (KNMI)](https://en.wikipedia.org/wiki/Royal_Netherlands_Meteorological_Institute). They provide some of their data sets as open data, however one will have to request an API key first.

## About
These have been the links I followed in order to find their open data product offerings.
- https://www.knmi.nl/over-het-knmi/about
- https://data.knmi.nl/
- https://www.knmidata.nl/open-data
- https://dataplatform.knmi.nl/catalog/

## Example
**Please note:** For realtime data download or downloading larger(bulk) parts of a dataset, you should use the (Open Data) API.
- https://dataplatform.knmi.nl/catalog/datasets/index.html?x%2Ddataset=Actuele10mindataKNMIstations&x%2Ddataset%2Dversion=2

## Open Data API
This looks really straight-forward. They even provide complete examples for Python, beyond what can be expected.
- https://developer.dataplatform.knmi.nl/get-started
- https://developer.dataplatform.knmi.nl/example-scripts

With kind regards,
Andreas.

pvgennip, 2020-11-26:
@amotl Thank you for your considerations, and self answering :-)
We currently load weather data for each hive that has lat/lon coordinates filled from the global weather API DarkSky (that has been recently taken over by Apple unfortunately, so this will discontinue end of 2021). We are looking into converting to another fair use global weather API.

pvgennip, 2020-11-26:
See :https://github.com/beepnl/BEEP/blob/11d6716b5451ee3747a22741a5c0f643bf772a0e/app/Http/Controllers/Api/WeatherController.php

amotl, 2020-12-03:
Dear Pim,

thanks for your pointers. Regarding another promising weather API, I would like to point out Bright Sky [1,2] by @jdemaeyer and contributors. However, it is unfortunately not a global weather API as it only works upon data by @DeutscherWetterdienst.

With kind regards,
Andreas.

[1] https://brightsky.dev/
[2] https://github.com/jdemaeyer/brightsky