beepnl/BEEP #69: Turn 400 errors into 500 errors

JuliaBD, 2020-12-01:
Such that Vue can access the returned error messages. Most importantly for the User endpoints.

pvgennip, 2020-12-03:
I will crete a branch vuejs-appv3 in which I will do this, since it will mess up the current AngularJS app.
@JuliaBD Do you also want 403 (not yours) and 404 (items not found) to be converted in 500s?

JuliaBD, 2020-12-03:
@pvgennip I think that's not needed, to begin with?
and could you please merge the v3-auth-export-adds with the vuejs-appv3 branch?

pvgennip, 2020-12-03:
Done

pvgennip, 2020-12-03:
Now also on-line at test server

amotl, 2020-12-03:
Dear Julia and Pim,

apologies for interrupting here. HTTP status codes actually have a real and important value for HTTP clients. So, I am humbly asking if you can share some more background about why Vue.js would not be able to read the body of HTTP 4xx responses?

HTTP status codes of 5xx clearly indicate server errors and we should not mess with them for a good reason. Please bear with me if I got something wrong on understanding this request. I will be happy to learn more about it.

With kind regards,
Andreas.

JuliaBD, 2020-12-07:
Dear Andreas,

Thank you for your reply, and you're absolutely right. We've now actually found a way to access the error messages for 4xx responses.

So I'm hereby reopening this issue, asking @pvgennip to revert this change, so we can go back to the correct HTTP status codes.

Kind regards,

Julia

JuliaBD, 2020-12-07:
@pvgennip please revert this change (see above message). In addition: could you please change the **invalid_password** message for incorrect passwords into **invalid_user** for the UserController?

pvgennip, 2020-12-14:
Acknowledged. No API change required.

closed 2020-12-14 by pvgennip

amotl, 2020-12-14:
Dear Julia and Pim,

thanks a bunch for listening and for adjusting back to the original version.

Keep up the spirit and with kind regards,
Andreas.