beepnl/BEEP #74: Fix typo in password-reset link

JuliaBD, 2020-12-01:
Current link:

https://appv3.beep.nl/password-forgotc39a2cea8c2d73256c32b7b1b02031fe311be1d647135cb9de5484814ec12/?email=j****@gmail.com

Should be:

https://appv3.beep.nl/password**-reset?code=**c39a2cea8c2d73256c32b7b1b02031fe311be1d647135cb9de5484814ec12**&**email=j****@gmail.com

pvgennip, 2020-12-03:
Done at test server

JuliaBD, 2020-12-17:
It has been changed to password-reset but there were more typos:

Current link:

https://appv3.beep.nl/password-resetc39a2cea8c2d73256c32b7b1b02031fe311be1d647135cb9de5484814ec12/?email=blabla@gmail.com

Should be:

-start with '?code=' after 'password-reset'
-'&email=' instead of '/?email='

https://appv3.beep.nl/password-reset?code=c39a2cea8c2d73256c32b7b1b02031fe311be1d647135cb9de5484814ec12&email=blabla@gmail.com

pvgennip, 2021-05-05:
Fixed on test

closed 2021-05-05 by pvgennip