beepnl/BEEP #80: Implement Alerts in API

JuliaBD, 2020-12-30:
**alert rule format:**
{
name: string,
description: string,
measurement: string, (physical quantity + unit (see sensor definitions input and output))
active hours: int array,
active months: int array,
period: int, (in minutes?) (select from minimum refresh time (retrieve from device) (in minutes?) or X, Y, Z hours)
number of times: int,
calculation: string, (min, max, av, der)
comparator: string, (=, <, >, <=, >=)
comparison function: string, (value, delta, abs value, abs delta)
value: float,
active: boolean,
}

**alert format:**
{
timestamp,
rule id,
alert function,
alert value,
read, (boolean)
}

**endpoints:**
GET predefined alert rules
POST alert rule
GET alert
PATCH alert (to toggle the read boolean)

pvgennip, 2021-02-09:
This will be implemented:
```
{
  id: int,
  name: string,
  description: string,
  measurement_id: int, (physical quantity + unit (see sensor definitions input and output))
  calculation: enum, (min, max, ave, der)
  calculation_minutes: int, (select from minimum refresh time (retrieve from device) or setting)
  comparator: enum, (=, <, >, <=, >=)
  comparison function: enum, (val, dif, abs, abs_dif)
  threshold_value: float,
  exclude_months: int array, (exclude, because a standard alert is 'always on') 
  exclude_hours: int array, (exclude, because a standard alert is 'always on') 
  alert_on_occurrences: int,
  alert_via_email: boolean,
  webhook_url: url, (optional endpoint to call for web hook integration)
  active: boolean
}
```

pvgennip, 2021-02-12:
Implemented on test server

rvaast, 2021-02-12:
Hi,

Where i can find the implémentation of this ?

pvgennip, 2021-02-12:
@rvaast in branch: https://github.com/beepnl/BEEP/tree/vuejs-appv3

rvaast, 2021-02-12:
Have you create an view for add rules or only use by api

JuliaBD, 2021-02-15:
@pvgennip please default the int arrays (exclude_hive_ids, exclude_months & exclude_hours) to empty arrays [] when not filled. Currently they contain an empty string [""], which does not work well with the select box.

@rvaast we're currently creating a view for this for the new version of the app (v3).

pvgennip, 2021-02-16:
@rvaast We are creating a VUE app in which we will create this view (see https://github.com/beepnl/beep-vue-app-legacy-api). Would you like to cooperate?

closed 2021-02-16 by pvgennip

rvaast, 2021-02-16:
@pvgennip if you want :) . I am not an expert in vue.js but i can watch. can you give me more details on the requirement (how do you expect things to be implemented) ? do you use slack or other to chat ?

pvgennip, 2021-05-05:
@rvaast that would be great, I will invite you to our BEEP Slack.

pvgennip, 2021-05-05:
Implemented at test and running evaluations