beepnl/BEEP #89: Docker compose does not find beep image

jzakotnik, 2021-02-21:
When running docker compose, this results in an error:
ERROR: pull access denied for beep, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

The image doesn't seem to be present on dockerhub, is it somewhere else?

Thanks!

pvgennip, 2021-03-29:
@joek Could you resolve this issue?

brtwrs, 2021-05-03:
@joek Did you fix this? Im getting the same error still

Thanks!

brtwrs, 2021-06-21:
Can anyone point me in the right direction please? :)

brtwrs, 2021-07-27:
I would really appreciate if someone can help us out😇 Would love to get started with BEEP!

amotl, 2021-07-28:
Dear @jzakotnik and @brtwrs,

maybe it already helps to build the image by invoking
```
docker build --label=beep:dev .
```

beforehand.

With kind regards,
Andreas.

jzakotnik, 2021-07-28:
Thanks for the update. Tried to build it myself in the past, but failed usually. This time it is
`executor failed running [/bin/sh -c ls -l && composer install && mv storage storage.bak && chmod -R 777 bootstrap/cache]: exit code: 255`

NewHopeApiary, 2022-01-20:
Also getting a similar error.  Wondering if anyone got it working with docker-compose.

C:\Users\winne\beep>docker-compose up
time="2022-01-19T19:04:46-06:00" level=warning msg="The \"PUSHER_APP_KEY\" variable is not set. Defaulting to a blank string."
time="2022-01-19T19:04:46-06:00" level=warning msg="The \"PUSHER_APP_CLUSTER\" variable is not set. Defaulting to a blank string."
[+] Running 0/3
 - beep Error                                                                                                      4.4s
 - mysql Pulling                                                                                                   4.4s
 - influxdb Error                                                                                                  4.4s
**Error response from daemon: pull access denied for beep, repository does not exist or may require 'docker login': denied: requested access to the resource is denied**

evlach, 2022-06-04:
I get exactly same error. Can anybody help?

lasicmk2, 2022-06-27:
The same here, anyone know how to solve the issue, please? Building image also doesn't work - https://github.com/beepnl/BEEP/issues/115