beepnl/BEEP #94: share hive in group as 'view only' is saved as 'editable'

JuliaBD, 2021-04-08:
When sharing a hive in a group, in both v2 and v3 the correct API call is being made when sharing a hive in 'view only' mode:
the hive_id is passed in the hives_selected array and NOT in the hives_editable array in the groups/{id} PATCH call

Expected behaviour when hive_id is NOT in the hives_editable array:
in the returned response, the group -> hives object contains the newly added hive with editable: false 

Actual behaviour:
in the returned response, the group -> hives object contains the newly added hive with editable: true.

Therefore, the hive is now wrongly 'editable' and being displayed as such.

JuliaBD, 2021-04-19:
Stephan heard about an issue the other way around: when provided editing rights, the result was view-only (in v2). But I can't reproduce this.

pvgennip, 2021-04-20:
@JuliaBD It looked like the hive in a group was saved with editable=true (while set to false), but the user that saved the hive  is an owner, and editable property was overrides by the ownership property. 

Now editable will always be false if the hive is set to editable=false

pvgennip, 2021-04-20:
Live on api.beep & test.beep

JuliaBD, 2021-04-20:
Same issue still occurs in v2 and v3:
Hives that are in hives_selected array but NOT in hives_editable array (with their hive id) should be saved in the group as 'view only' (editable: false), but they keep being saved as 'editable: true'

pvgennip, 2021-05-05:
I see this solved in v2.2.1 of the app on the test and live server. All hives defined as 'editable' false (eye icon) are saved as 'editable':false, which is represented in the result of https://test.beep.nl/api/groups.

closed 2021-05-05 by pvgennip