beepnl/BEEP #106: Not displaying all received data on app.beep.nl

wasalm, 2021-05-28:
Dear Beep,

At home we have our own monitoring system and we publish our data to app.beep.nl
Due to the restrictions in the Lora protocol we send our data in two lora messages.
The first contains the weight and temperature of our beehive and the second only the audio data.
This used to work, but recently the beep app displays only the graphs of the latest message.
That is, if the last message contains the weight, I see graphs with the weight, but the graphs with the sound information is gone. Also vice versa. If the last message contains the audio data, I only see the graph with the audio data and not the graphs with the weight.
Can you please revert your changes and show both graphs at the same time?

Best wishes,

Andries

pvgennip, 2021-05-28:
Hi Andries,

Excuse me, I indeed see a bug in the #106.
I will add some extra logic to include both graphs

Kind regards,
Pim van Gennip

pvgennip, 2021-05-28:
Issue should be solved

closed 2021-05-28 by pvgennip