beepnl/BEEP #108: alerts triggered too often

JuliaBD, 2021-06-01:
Alertrule: minimum waarde van gewicht < 10kg, elke 15 minuten direct (zie screenshot).

**Measurement Data:**
14 mei Kastv3 (BEEPBASE-TESTv3)
13:15u: 20kg
13:30u: 13.55kg
13:45u: 0.00kg
14u: -0.01kg

**Alerts:**
vanaf 14 mei 23:36u, elke 2u en 16 minuten:
alert_value: -0.01kg (en schommelt daarna tussen 0.00 en -0.03) (behalve 18-31 mei hiaat, wrsch ivm alert gedeactiveerd?)

Zelfde issue voor andere alert rule en hive (Test 5 hive: batterijvoltage gemiddelde waarde > 2.1v kwam elke 2u en 11 minuten (bij waardes zoals 3.30, 3.26, 3.27 etc.))

**2 issues:**
1. Na de eerste melding zou er geen alert meer gegenereerd moeten worden?
2. Er gaat iets mis met de tijdstippen: 23.36u van de eerste melding lijkt niet te kloppen / 10u te laat?

**1 vraag:**
Alert.created_at heeft (bijv.) format '2021-05-14 12:00:00', measurements[0].time heeft format '2021-05-14T12:00:00Z' (=utc, =14u NL tijd). Kloppen de created_at tijden van de alert al met de timezone of moeten ze nog worden omgezet? Maw: misschien beter beide in utc format om het gelijk te trekken? Of kan ik ze tonen as is (alleen leesbaar gemaakt door moment)?

[image]

pvgennip, 2021-06-29:
@JuliaBD `created_at` is created by the server and will always be in UTC. So please convert this date to the applicable timezone of the user (just like inspections)

JuliaBD, 2021-06-29:
@pvgennip With inspections, for a POST or PATCH call the users date+time is posted, f.e. date:"2021-06-29 14:19", which is returned with a GET call as: created_at:"2021-06-29 14:19". (Both in v2 and v3).

In other words: the created_at value for an inspection is NOT converted to utc (also, btw, timezone is not part of the POST/PATCH request).

For alerts, I can treat the created_at value as utc and convert it to the users timezone, but then we have to be sure it really is in utc? Seeing issue 2 something seems to be wrong with the timing anyway, so it's a bit tricky to evaluate now.

pvgennip, 2021-09-20:
Now all times are in UTC at `test.beep.nl` and alerts should be working correctly.

closed 2021-09-20 by pvgennip