beepnl/BEEP #109: Log feeding_box & queen_excluder layer in hive action inspection

JuliaBD, 2021-06-01:
In v3 when placing a feeding box and queen excluder, a hive action is added as an inspection, but this only mentions the number of brood and honey layers and frames (see screenshot). This is confusing as nothing happened with those layers.

Solution:
Feeding box and/or queen excluder should be added as inspection items for the 'hive action' inspection when adding or removing one or more of those layers.

Please note that there can now be multiple queen_excluder layers (and only 1 feeding box layer)

[image]

pvgennip, 2021-08-31:
Added at test.beep.nl

closed 2021-08-31 by pvgennip

pvgennip, 2021-08-31:
Works for v2 app too