beepnl/BEEP #110: in v2: hive brood box depth is saved as height and vice versa

JuliaBD, 2021-06-03:
ng-model of bb_depth_cm and bb_height_cm are mixed up in hive_edit.html, which results in what user submits as brood box height is saved as brood box depth, and vice versa.

In v3, depth and height are correctly displayed, BUT if the hive dimensions were configured in v2, the wrong values are shown according to the users intentions (what the user submitted in v2 as height is shown (because saved as such, see above) as depth and vice versa).

Maybe best fixed via a database query? Or add a note to 'New' / support?

pvgennip, 2021-08-31:
I will rename the database columns and adjust the v2 app accordingly

pvgennip, 2021-09-28:
Fixed in commit bd61533

closed 2021-09-28 by pvgennip