beepnl/BEEP #112: posting of historic data that are collected while no internet available

Snooz82, 2021-06-07:
Hi Beep Team,

I am a user of Beep (with Bee-Observer) and i have the case that we have here and then locations where the internet access is not very stable or where no internet at all is available.
Therefore we have the possibility to store the measured data together with a timestamp on an SD-Card.

Now the question would be: 
Would you see any possibility to transmit multiple measurements with a specified timestamp instead of putting the current sensor values with the current time in the database?

Like when the internet is off for a day we would not lose these scientific data.

pvgennip, 2021-06-08:
@Snooz82 That is possible, please check the API documentation at api/sensors in https://api.beep.nl/docs/

closed 2021-06-08 by pvgennip

Snooz82, 2021-06-08:
@pvgennip 
It would be nice to not just close the issue without giving any new information.

POST api/sensors has just the data field.

how can i set the timestamp for a specific measurement?
Can you please add this information. And in which format need this to be?

thanks.

Snooz82, 2021-06-08:
I have now found in the code (not in the api docs) that a key `time` In the measurement data allows to set data to a specific time.

Still not the solution, because beep.nl has a limitation of requests per time.

if i try to send multiple measurements, i got blocked due to too many requests.

actually i would see a POST api/sensors_history that accepts a list of measurement objects as very helpful.
Then one single call could send all data since last connected/transmitted measurement.

ClemensGruber, 2021-06-10:
Hi Pim / @pvgennip we have discussed this issue also on https://community.hiveeyes.org/t/daten-puffern-und-als-bulk-an-beep-bob-schicken/4037/ in case you have some time for a look. With the current implementation via `value_key "time"` we faced to two limitation while (historic / buffered) mass data ingest:

1. We collect data in a high frequence, so we have many many datasets per hour / day. By ingesting this amount of data we noticed a throttle mechanism at beep that limits calls: 

https://github.com/beepnl/BEEP/blob/2ebb780f300e8cfd371e3eb2e52d9c7a30fa6fa9/routes/api.php#L23-L28

In case we know what `throttle:10,1` means and how we can mitigate it we could solve this problem vor BOB / bee-observer.org

2. But even we can "disarm" the throttle mechanism we have still many calls. @Snooz82 calculated around 720 calls for one hour data. I know this is not necessary for a "normal" beekeeper but it was a wish from our scientific approach. 

So our question would be: Is there a mechanism where we can ingest lets say 100 datasets and not just one per call? In case mass ingest is not possible in the current software implemetation. What is the best way to get as many POSTs as possible in the database, e.g. changing the throttle values?