beepnl/BEEP #116: All date/times in utc format

JuliaBD, 2021-08-17:
For @pvgennip: make sure all terms are saved/returned in utc format. If timezone is present in header, use that together with the provided local date/time to convert to utc.

For @JuliaBD: make sure timezone is provided in header esp. when posting/patching a data/time var. And make sure all returned date/times are converted to users timezone when displayed.

pvgennip, 2021-08-31:
Manual input (i.e. inspections/photos/apiary/device/hive/etc) would need the timezone header to store the time in UTC. It requires a lot of refactoring/adjustment/database transformations/roleout timing to get this done correctly

JuliaBD, 2021-09-02:
Or, for devices and sensor definitions, keep it as is, as otherwise database values need to be updated?

pvgennip, 2021-09-17:
All times at test.beep.nl are now in UTC.  date() was using Europe/Amsterdam time, now UTC (since config/app.php has UTC as default timezone)

pvgennip, 2021-09-20:
@JuliaBD please check if all times are ok now

closed 2021-09-20 by pvgennip

JuliaBD, 2021-09-20:
Yes!