beepnl/BEEP #132: add decline group invitation option to /groups/checktoken POST request

JuliaBD, 2021-11-22:
It's currently not possible to decline a collaboration group invitation, please add this option to the API.

Perhaps add a non-required decline prop to the /groups/checktoken POST request that needs to be true in order to decline the invitation instead of accept it, which removes the invitation from the list.
For accepting, this prop is either not present or false.

pvgennip, 2021-11-22:
Implemented on test.beep.nl/api

property `decline` can be `true` or `false` and is default `false`

JuliaBD, 2021-11-23:
I did a /groups/checktoken POST request with payload:

{"group_id":62,"token":"m6FBLABLA","decline":true}

But this resulted in an accept... (200 response, message: "group_activated". And the group appears in the overview.)

Two questions:
1. Could you please make sure decline: true results in a decline (with appropriate response message).
2. Would it be a good idea to give the declining user a value 'declined' for the user.invited prop, to provide feedback for the group admin (in the group details table)? Or will the user be removed altogether?

pvgennip, 2021-12-24:
Fixed

JuliaBD, 2021-12-27:
I still experience the first issue, above, that a checktoken request with decline: true results in a 200 'group_activated' response:

I did a /groups/checktoken POST request with payload:

{"group_id":62,"token":"m6FBLABLA","decline":true}

But this resulted in an accept... (200 response, message: "group_activated". And the group appears in the overview.)

(And therefore haven't been able to check whether the second point has been implemented correctly yet)

pvgennip, 2021-12-27:
Sorry, fixed now

closed 2021-12-27 by pvgennip