beepnl/BEEP #198: POST api/sensors replies with 500 error "sensor-write-error" (self-hosted)

schreu, 2024-10-21:
Upon sending data through the API I am getting a 500 error.
After investigations it seems that there is an issue in the communication between the app and the InfluxDB (401 unauthorized).

pvgennip, 2024-10-21:
Hi @schreu 
It might have to do with:
- InfluxDB database not being created 
- An InfluxDB database user not being created 
- The InfluxDB database user credentials not being updated in your .env settings

Please follow all steps in the Readme file

closed 2024-10-21 by pvgennip

schreu, 2024-10-21:
Hi @pvgennip ,

Thank you for your feedback.
The issue came from InfluxDB2 requiring a token to authenticate instead of user:pass.
I solved it with the below command to be ran with docker exec -it influxdb bash:
influx v1 auth create --username user_influx --read-bucket [your bucket id] --write-bucket [your bucket id]

probably valid only for new installations that start from 0 with an Influx2 DB and worth mentioning in the Readme file

(and by the way, nice work on BEEP ! )