beepnl/BEEP #199: Lora weight data not stored

antooni007, 2025-01-30:
Hi, 

I'm trying to log sensor data over TTN to the lora_sensors api endpoint. While trying out the test payloads that are stated in the ttn-payload-decoder.js file, I noticed that all sensors are correctly logged on the beep app, except for the weight. It does not seem to store the weight data anywhere.

E.g., payload 

1B0C4B0C44640A01012D2D040107D6

Stores the correct battery voltage (3.14V) and hive temperature (20.06 C), but the weight data is just ignored. To my understanding, a weight of 77.101 kg should be read from this payload. (The data is sent to port 3)

Is this a known issue or am I overlooking something? Thanks in advance for your response.

Kind regards,

ezja, 2025-02-01:
@antooni007 Are you posting your weight values in the `w_v` field? If so, these values will displayed only after they have been converted to `weight_kg` which is normally done by the following Sensor Definition:

[image]

When using the BEEP Base hardware, the Bosche H40A load cell is calibrated with the BEEP Base App and this results in the "Offset" and "Multiplier" constants needed to calculate `weight_kg` values from the H40A's raw `w_v` values.

If you are using different scale hardware and you already know your calibrated weight values, simply post them to the `weight_kg` field, or create a new Sensor Definition on the app's Devices page that has input `w_v` and output `weight_kg` with offset = 0 and multiplier = 1.

antooni007, 2025-02-01:
Hi @ezja 

I am indeed posting to the `w_v` field from my own scale hardware. But even when I create the sensor definition as you showed,
[image]
I don't get any weight data from the payload I sent it:
[image]

Marchellos, 2025-03-03:
And.. Has the problem been solved in the meantime? 
I have that with "h_i", for example.
"h" works again without any problem.

antooni007, 2025-03-03:
Nope, still having the same issue...

pvgennip, 2025-03-03:
Could you post the full payload that your system sends? @Marchellos and @antooni007

pvgennip, 2025-03-03:
You can test the JSON results of a LoRa payload by getting this URL with e.g. Postman:
https://api.beep.nl/api/sensors/decode/p/{port}/pl/{HEX payload}

Marchellos, 2025-03-03:
"h' does come in; " h-i" I don't see.
Just removed it again, and put it in again.

https://api.beep.nl/api/sensors?key=xxxxx&h=36.7&h_i=36.5"

antooni007, 2025-03-04:
The problem I'm having is that it specifically doesn't work for the lora_sensors endpoint of the api, using the TTN webhook. This should allow me to forward lora data to the platform without the need of a middle man web server?

The webhook settings: 

[image]

This is the full payload forwarded from TTN: 

_{
  "name": "as.up.data.forward",
  "time": "2025-03-04T17:10:44.910397740Z",
  "identifiers": [
    {
      "device_ids": {
        "device_id": "beehub2",
        "application_ids": {
          "application_id": "beehives1"
        },
        "dev_eui": "70B3D57ED006DBB8",
        "join_eui": "70B3D57ED0028D38",
        "dev_addr": "260B51AC"
      }
    }
  ],
  "data": {
    "@type": "type.googleapis.com/ttn.lorawan.v3.ApplicationUp",
    "end_device_ids": {
      "device_id": "beehub2",
      "application_ids": {
        "application_id": "beehives1"
      },
      "dev_eui": "70B3D57ED006DBB8",
      "join_eui": "70B3D57ED0028D38",
      "dev_addr": "260B51AC"
    },
    "correlation_ids": [
      "gs:uplink:01JNH26W6Q7WY1MHCQSB8J7MEM"
    ],
    "received_at": "2025-03-04T17:10:44.903300192Z",
    "uplink_message": {
      "session_key_id": "AZViIlXGMAldnpcjEk41sg==",
      "f_port": 3,
      "f_cnt": 1,
      "frm_payload": "GwAAD61ZCgEBLS0EAQct",
      "rx_metadata": [
        {
          "gateway_ids": {
            "gateway_id": "eui-00005813d35e2d50",
            "eui": "00005813D35E2D50"
          },
          "time": "2025-03-04T17:10:26.670Z",
          "timestamp": 943020724,
          "rssi": -120,
          "channel_rssi": -120,
          "snr": -9.8,
          "uplink_token": "CiIKIAoUZXVpLTAwMDA1ODEzZDM1ZTJkNTASCAAAWBPTXi1QELS11cEDGgwIlOicvgYQ69muywIgoJ6pg7nKGioMCILonL4GEIDHvb8C",
          "channel_index": 3,
          "gps_time": "2025-03-04T17:10:26.670Z",
          "received_at": "2025-03-04T17:10:44.682488759Z"
        }
      ],
      "settings": {
        "data_rate": {
          "lora": {
            "bandwidth": 125000,
            "spreading_factor": 9,
            "coding_rate": "4/5"
          }
        },
        "frequency": "867700000",
        "timestamp": 943020724,
        "time": "2025-03-04T17:10:26.670Z"
      },
      "received_at": "2025-03-04T17:10:44.696269047Z",
      "consumed_airtime": "0.226304s",
      "network_ids": {
        "net_id": "000013",
        "ns_id": "EC656E0000000181",
        "tenant_id": "ttn",
        "cluster_id": "eu1",
        "cluster_address": "eu1.cloud.thethings.network"
      }
    }
  },
  "correlation_ids": [
    "gs:uplink:01JNH26W6Q7WY1MHCQSB8J7MEM"
  ],
  "origin": "ip-10-100-5-252.eu-west-1.compute.internal",
  "context": {
    "tenant-id": "CgN0dG4="
  },
  "visibility": {
    "rights": [
      "RIGHT_APPLICATION_TRAFFIC_READ"
    ]
  },
  "unique_id": "01JNH26WDET7CRHCF3Y7S25929"
}_