beepnl/BEEP #203: After creating hive change (inspection) - can not change details due to 404

crunchie84, 2025-08-28:
I'm making some backdated notes in my beep app administration. One of these is the changing of the hive configuration. After making the changes to the hive it gets recorded with datetime = `now()`. Want to change the date to the actual date. Get a 404 not found for the inspection details / edit page.

### reproduction
- change hive configuration (=inspection data object)
- show list of inspections of hive -> shows inspection with description 'kast handeling' (check: inspection is saved correctly)
- press ´edit inspection' button
- page shows 404 inspection not found

### api endpoints
GET `https://api.beep.nl/api/inspections/hive/49425?page=1`
``` 
{
    "inspections": {
        "current_page": 1,
        "data": [
            {
                "id": 430587,
                "notes": "Kast handeling",
                "reminder": null,
                "reminder_date": null,
                "impression": null,
                "attention": null,
                "created_at": "2025-08-28 21:17:00",
                "checklist_id": null,
                "image_id": null,
                "owner": true,
                "thumb_url": null,
                "hive_id": 49425,
                "location_id": 5152,
                "item_count": "5",
                "searchable": []
            },
            {
                "id": 430586,
....
``` 

details - GET `https://api.beep.nl/api/inspections/430587`
response HTTP 404 = `{}` as body

pvgennip, 2025-09-04:
@crunchie84 thanks for the detailed error message. I am currently solving the issue.

pvgennip, 2025-09-04:
@crunchie84 the issue, which was a cache race condition, has been solved in production.

closed 2025-09-04 by pvgennip

crunchie84, 2025-09-16:
Checked and verified, Thanks! 🥳